foxglove/studio #1837: Diagnostics Panel: Allow filtering diagnostics by level

jessicaaustin, 2021-09-11:
I'm interested in adding a new setting for the Diagnostics panel that allows filtering diagnostics by level. 

This would allow me to create a layout with two diagnostics panels: one that has only "bad" diagnostics, and one that has all diagnostics. This separation reduces cognitive load for the user -- they can look in one place for everything that's going wrong, and ignore the rest.

Note: This is similar to how rqt Robot Monitor works:

[image]

jessicaaustin, 2021-09-11:
See https://github.com/foxglove/studio/pull/1838 for an example of me playing around with this concept. 

Here, I've added a simple boolean to check if a diagnostic is OK or not. 

Probably having checkboxes to show/hide each level makes more sense?

closed 2021-09-16 by esthersweon

jtbandes, 2021-09-16:
Thanks for your interest and contribution! We've implemented a more detailed level filter in https://github.com/foxglove/studio/pull/1877 that will be in the next release. Let us know what you think 😃

jessicaaustin, 2021-09-16:
@jtbandes Nice, I like it! I pulled down `main` and tried it out against one of our bag files, and was easily able to split out "bad" diagnostics from everything else.