foxglove/studio #1840: Remove backfillDuration from seekPlayback api

defunctzombie, 2021-09-11:
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/types.ts#L77

Player seek api should only accept a seek location. The amount to backfill as an internal setting - maybe one day exposed during player construction.

Related: Look into [this comment](https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/OrderedStampPlayer.ts#L172-L174) in OrderedStampPlayer which seems to imply that backfill does a forward fill? Seems like another abuse of the argument and a reason to internalize to players what happens when seeking.

jtbandes, 2021-09-11:
> backfill does a forward fill?

The whole point of OrderedStampPlayer is to reorder messages, so this doesn’t seem like a terribly strange thing or abuse….

defunctzombie, 2021-09-11:
I agree that OrderedStampPlayer may have a different "backfill" strategy but I don't think this makes sense as part of the public player API. If RandomAccessPlayer wants to expose an API for OrderedStampPlayer use we should do that between those two players - right now there is already some implied coupling (https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/RandomAccessPlayer.ts#L548).

This is related to my continued dislike of the sort of pseudo-abstract pipeline/wrapping that happens between some players as well as randomAccessDataProviders. Each part appears "separate" and transposable but actually has been designed to sit in a specific place.

jtbandes, 2021-09-13:
> right now there is already some implied coupling (https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/RandomAccessPlayer.ts#L548)

The comment does mention OrderedStampPlayer by name as motivation, but otherwise it doesn't seem like real coupling. backfillDuration is just a parameter to seekPlayback that OrderedStampPlayer happens to set. This seems like a normal way for separate components to work together 🤷

defunctzombie, 2022-03-14:
The new IterablePlayer currently ignores the backfill duration tho I think it should probably implement that feature for supporting the OrderedStampPlayer in header playback mode.

defunctzombie, 2022-03-14:
Closing this issue for now. I don't think we will tacking it individually.

closed 2022-03-14 by defunctzombie