foxglove/studio #1841: ROS2 direct connection is not established

alluring-mushroom, 2021-09-13:
**Description**
I have ROS2 nodes running. Foxglove detects that they are there, but cannot see the topics they are publishing

- Version: 0.17.0
- OS: Ubuntu 20.04.3 (kernel: Linux 5.11.0-27-generic)

**Steps To Reproduce**

1. Start foxglove.
2. Start a ROS2 node. I used demo_nodes_py talker, but also tried `ros2 topic pub /test std_msgs/msgs/String`)

**Expected Behavior**
I should be able to see the topics in topic browser, raw messages panel, etc.

**Actual Behavior**
Foxglove detects the ros2 nodes, which I can see in the dev tools (see screenshot) but I cannot observe the result in any Foxglove panels.

The nodes work over the ros2 webbridge, and the ros1 direct connection with ros1-ros2 node running.

[image]

alluring-mushroom, 2021-09-13:
Note: the websocket connection in the dev-tools is just me changing datasource back and forth, attempting to get foxglove to detect the ros2 nodes

amacneil, 2021-09-13:
@alluring-mushroom can you test this in v0.18.0 which was just released?

alluring-mushroom, 2021-09-14:
Yep, I've updated and tested with the same options (talker node, straight `ros2 topic pub` with a String. Appears exactly as before unfortuneately

sink0918, 2021-09-16:
I have tested with v0.18.1 using talker node with ROS2 connection, also can't receive any messages in Studio.
OS: Windows 11, remote machine: Ubuntu 20.04

[image]

amacneil, 2021-09-16:
Thanks for reporting this. Our ROS 2 lead is going to be ooo for the next couple weeks, so in the meantime are you able to use the ROS 2 rosbridge connection?

jtbandes, 2021-09-16:
@sink0918 When using Foxglove Studio to connect to a **remote** machine, we recommend using the rosbridge connection, because the direct connection has more unusual networking requirements such as UDP multicast.

@alluring-mushroom If possible, please share more detailed steps to reproduce the issue. I've tried demo_nodes_py/cpp talker and it works for me:

- Foxglove Studio 0.18.1
- Ubuntu 20.04 (running in a Parallels VM)
- ROS 2 Galactic

https://user-images.githubusercontent.com/14237/133661999-69264772-21f6-46e5-8d5a-7a4469fc298a.mov

sink0918, 2021-09-17:
@jtbandes When connected to __local__ machine on Windows, Foxglove Studio can't work either.
- Foxglove Studio 0.18.1
- Windows 11
- ROS 2 Foxy

alluring-mushroom, 2021-09-17:
@jtbandes
I have attached a recording in a similar manner to you.
If you need any specific information, let me know

@amacneil I have been happily using the rosbridge connection in the meantime, much easier than the ros1 bridges I was using before

- Foxglove 0.18.1
- Ubuntu 20.04 running on metal
- i3wm (if that affects anything, it would be unfortunate)
- ros2 foxy 

https://user-images.githubusercontent.com/86041465/133724625-c4f4b0de-3290-4da4-a9ae-3dec254a23a0.mp4

jtbandes, 2021-09-17:
Most of our testing so far has been on Galactic. I would guess these issues are happening because both of you are using Foxy. We can work on improving Foxy support, but in the meantime, you can try with Galactic as well let us know how that goes.

alluring-mushroom, 2021-09-23:
@jtbandes I've tested on galactic, and it indeed works. We would prefer to use LTS versions of ROS, so foxy support would be ideal. 
Thank you so much for your help :smile:

jhurliman, 2021-10-06:
Our underlying RTPS library https://github.com/foxglove/rtps had not been tested with FastRTPS, the default DDS/RTPS transport for ROS2 Foxy. (Galactic uses Eclipse Cyclone DDS.) I've made a number of fixes in https://github.com/foxglove/rtps/pull/9 that should make that library Foxy-compatible. I'll close this ticket once a new version of the relevant libraries have been published and studio is updated.

closed 2021-10-07 by jhurliman