foxglove/studio #1868: URL params for connection and layout

amacneil, 2021-09-15:
I should be able to set the connection (player + params) and the layout from URL query params. These should update in real time as I use the app, so that the current URL can always be copied.

defunctzombie, 2021-09-15:
The user-story here is a desire to open the web version of studio with a specific player and parameters. This won't work for all players. We need to be careful about how we introduce/surface some of these concepts and trying to url/browser specific things deep within the app. This was something that I had to extract from PlayerManager and I'd like to avoid us putting it back in the same way since we have web and desktop.

This is also related to our data platform player and `foxglove://` url handling for desktop.

amacneil, 2021-09-15:
Good point - might be worth separating "layout in the url" from "connection details in the URL".

- Layout makes sense for shared layouts, and personal synced layouts, but unclear how we want to handle localstorage layouts? Maybe it is ok if you save a personal or localstorage layout URL and that URL only works on your computer, since other people don't have access to it?
- Connection details
  - Fairly straight forward for remote connections (ros1+2, rosbridge, http bag, foxglove DP)
  - Maybe also ok for file based connections? We can just have an encoded file path in the URL? Still nice that I can bookmark something and come back to it.

There is a separate discussion about how we can make these URLs "shareable" within the desktop build of Studio (Linear is a good example of prior art, Slack also has some "copy link" options although at per-message level rather than app level).

defunctzombie, 2021-09-15:
Related: https://github.com/foxglove/studio/issues/1848

defunctzombie, 2021-12-03:
Re-assigned to @miles who's actively working on this in https://github.com/foxglove/studio/pull/2171

closed 2021-12-06 by foxymiles

miles, 2021-12-13:
How is it my username being used on this project? I keep getting notifications on it.

foxymiles, 2021-12-13:
@miles I think @defunctzombie meant to tag me, @foxymiles .

defunctzombie, 2021-12-13:
@miles Yea - sorry about that - the username autocomplete led me astray.