foxglove/studio #1878: macOS automatic updates are broken

amacneil, 2021-09-16:
**Description**
via @defunctzombie:

```
Checking for macOS Rosetta environment
Checked for macOS Rosetta environment (isRosetta=false)
updater cache dir: /Users/roman/Library/Application Support/Caches/foxglove-studio-updater
Update has already been downloaded to /Users/roman/Library/Application Support/Caches/foxglove-studio-updater/pending/foxglove-studio-0.18.0-mac-universal.zip).
Creating proxy server for native Squirrel.Mac (fileToProxy=https://github.com/foxglove/studio/releases/download/v0.18.0/foxglove-studio-0.18.0-mac-universal.zip)
Error: TypeError: Cannot read property 'port' of null
    at O (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37873)
    at n.updateDownloaded (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37959)
    at Object.done (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37457)
    at V (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:29470)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async n.executeDownload (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:29598)
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
TypeError: Cannot read property 'port' of null
    at O (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37873)
    at n.updateDownloaded (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37959)
    at Object.done (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:37457)
    at V (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:29470)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async n.executeDownload (/Applications/Foxglove Studio.app/Contents/Resources/app.asar/main/main.js:67:29598)
```

- Version: Since around v0.17?
- OS: macOS

defunctzombie, 2021-09-16:
https://github.com/electron-userland/electron-builder/pull/6149

jtbandes, 2021-09-16:
Tested in dev mode by creating .webpack/dev-app-update.yml with:
```yml
provider: github
owner: foxglove
repo: studio
```
Confirmed that with electron-updater@4.4.5, I see the port error in console. With 4.3.9, which I accidentally downgraded to in https://github.com/foxglove/studio/pull/1776/files, the error is not present. With 4.6.0, the version we are currently using as of https://github.com/foxglove/studio/pull/1849, the error is also not present.

closed 2021-09-16 by jtbandes