foxglove/studio #1891: Touch screens not supported (?)

adityakamath, 2021-09-18:
I use Foxglove Studio on a laptop which has a touch screen, but I am unable to use Foxglove with it. I am forced to use a mouse (which means I cannot carry my laptop around behind my robot). I am not sure if it is an issue with my laptop or are touch screens not supported in general?

If its the latter, would love to see touch screen support in the future..

- Version: Foxglove Studio v0.18.1
- OS: Windows 10
- Laptop: Asus Zenbook Flip 14

jtbandes, 2021-09-19:
Hi, thanks for the feedback! Can you please share some more details about the problem? For example: which actions are you unable to do using the touchscreen, which ones are possible but difficult, and which ones work well?

adityakamath, 2021-09-22:
Hi @jtbandes, to elaborate, I am able to click and select things on the Studio dashboard, but unable to rotate/zoom into the 3D view using the touch screen.

jtbandes, 2022-09-08:
Done in #4011. We can file separate tickets to track other touch screen compatibility features if desired. A lot of basic interactions do work today, and we aren't currently planning to put a lot of effort toward adding more touchscreen support.

closed 2022-09-08 by jtbandes

adityakamath, 2023-03-17:
I accidentally realized that touchscreen input was working (while getting some dust off my screen), and came back to this. The basic interactions are just what I needed, thanks for the update! 😄