foxglove/studio #1914: Extensions API: Add ability to open sibling panels

jessicaaustin, 2021-09-28:
Originally posted at https://github.com/foxglove/fox/issues/28

Background:

I'm building an extension that shows data from a topic, and I would like to add the ability to open a sibling plot to show values over time, similar to the functionality in the Diagnostics panel and Raw Messages panel (they call `openSiblingPlotPanel` and `openSiblingStateTransitionsPanel` from `Plot`). 

Request:

Please add a function to the `PanelExtensionContext` to allow opening a sibling panel, with all the necessary props.

Thank you!

jtbandes, 2021-10-29:
I believe the displayName issue is due to a typo here: https://github.com/foxglove/studio/blob/44212a8aaab9896b5ae30bf212c5eb3234034890/desktop/renderer/providers/ExtensionLoaderProvider.tsx#L25

Resolving this might be a bit tricky because we don't want to break existing layouts that were depending on this behavior :confused:

The initialState issue you described seems like it might be a side effect of the fact that we open the new panel and _then_ add the new panel's state: https://github.com/foxglove/studio/blob/44212a8aaab9896b5ae30bf212c5eb3234034890/packages/studio-base/src/components/Panel.tsx#L419

@defunctzombie Is there a way we can tell the extension panel when its state has changed? Currently it looks like the panel is given initialState, but in the future it's not able to receive state changes. This would also affect the `updateIfExists` behavior for addPanel.

defunctzombie, 2021-10-30:
> Resolving this might be a bit tricky because we don't want to break existing layouts that were depending on this behavior

I think we need to create a transition period and resolve this. Its only going to become even more confusing over time.

> I struggled a little bit trying to figure out what I should use for the `type` name;

We could list the fully qualified type under the panel name in the add panel list? It's not ideal since for most users that information is un-actionable and meaningless.

Open to ideas here - ideally something that users don't need to normally see or interact with. Maybe under the extension details (when you click a specific extension) we can show the panels that extension adds. Its a bit of a catch-22 with your own extension tho.

> Is there a way we can tell the extension panel when its state has changed

We could add another field on the renderState - though we need to be careful about the extension calling `saveState` triggering another render update with the same state maybe?

We could have the user register another handler similar to renderState that receives updates. The design I was aiming for  with the extension api was that there was only one function that would be called for the panel extension with updates but maybe it's ok to break this down between "render" updates from data arriving and other user-triggered events like state updates?

jessicaaustin, 2021-10-30:
My two cents as a user on figuring out what to put for `type`: just add an error log message if the type I pass in isn't valid ([currently it fails silently](https://github.com/foxglove/studio/blob/c36bb326d71d752ac78a115d44a03911bd9e9515/packages/studio-base/src/components/Panel.tsx#L382)). Someone calling this method is going to be a developer anyway, so if something goes wrong the first thing we'll do is open up the dev console and look for errors.

jessicaaustin, 2021-11-02:
Is it possible to add a ticket for the initial state issue as well? At this point, this `addPanel` functionality does not work for panels that use the extension api.

foxymiles, 2021-11-30:
It looks like what's happening here is that studio is registering the panel "type" as a concatenation of the source package and panel name:

https://github.com/foxglove/studio/blob/4141b49b7a0fb3c212c75c39613a457a969866dd/packages/studio-base/src/providers/PanelCatalogProvider.tsx#L38-L40

In my case I've created a package called `fg-ext` and a detail panel called `Miles Detail Panel`. In my main panel I'm able to open the detail panel successfully with this code:

```
    context.layout.addPanel({
      position: "sibling",
      type: "fg-ext.Miles Detail Panel",
      updateIfExists: true,
      getState: (existingState?: unknown) => {
        console.log(existingState);
        return {
          topicPath: "/diagnostics",
        };
      },
    });
```

@jessicaaustin Does it work for you if you try this approach?

jessicaaustin, 2021-11-30:
@foxymiles I am able to open an extension panel as a sibling panel. However the state does not get populated. So in your example that would mean `topicPath` would not get set on the `initialState` in your `Miles Detail Panel`. Is that working for you?

If I restart Studio or switch layouts and then come back so that the panel loads state from storage, then the state I requested is there. So the bug is just that when the sibiling panel is initially created; the initialState is not set as provided in the call to `addPanel`. See the comment here: https://github.com/foxglove/studio/issues/1914#issuecomment-955061122

Just verified this is still an issue with the latest version (0.22.1).

Thanks!

foxymiles, 2021-11-30:
@jessicaaustin Yes you're right populating the initial state is a separate issue. Sorry if my comment wasn't clear on that point. I'm still looking into possible fixes for the state problem.

jessicaaustin, 2021-11-30:
Ah no worries, I just misread your comment. Thanks for looking, I am happy to help test or provide feedback for any fixes!

foxymiles, 2021-11-30:
@jessicaaustin I'm not sure if it's easy for you to run studio from a branch there but if so this branch seems to fix the panel initial state problem for me. https://github.com/foxglove/studio/pull/2232

PR #2232 (2021-11-30): Initialize extension panels with the correct default state.
**User-Facing Changes**
This fixes an issue in the extension panel API.

**Description**
This makes sure that when sibling panels are added via the extension panel API's `addPanel` function that they are initialized with the correct state.

Fixes #1914