foxglove/studio #1917: Better handling for API outages

amacneil, 2021-09-29:
Currently if Studio encounters an error talking to API (server down or other connectivity problem) we get toast notifications:

[image]

We should rethink the UX here and maybe display some notification indicator or other element that indicates we are currently disconnected, without spewing big errors like this.

cc @2metres for thoughts

defunctzombie, 2021-10-22:
I think we already detect network state - these are separate from being offline and happen when the server responds with an error. @jtbandes I believe you recently made a change to limit how many of these appear?

@amacneil besides not having many copies of the same error appear - what else do you want to see for this issue? Do you want the error localized to the layout sidebar? Something else?

defunctzombie, 2022-04-22:
I think we've wrapped more of these errors now - at least I don't find myself seeing these anymore. I was not able to reproduce these errors when I turned my wifi off on my laptop.

@amacneil are you able to reproduce? If not lets close this out and re-open when we are able to.

amacneil, 2022-04-22:
I did an experiment opening studio in one tab and logging out in the other tab. It gave me one toast and then logged out, which seems not super polished but ok for now. I haven't seen toast spam like this in a while so fine with closing this ticket.

closed 2022-04-22 by amacneil