foxglove/studio #1922: Improve built-in support for commonly used messages

jtbandes, 2021-10-01:
Tracking some well-known message types which we could consider adding built-in Studio panels/visualizations for.

- https://docs.ros.org/en/melodic/api/sensor_msgs/html/index-msg.html 
- https://github.com/ros-planning/moveit_msgs
- https://github.com/luxonis/depthai-ros/tree/noetic-devel/depthai_ros_msgs/msg

- #1628
- #1629
- #1630
- #1325
- #1232
- #727
- #1049
- #1615

wkalt, 2022-04-13:
the tutorial here also allows easy interaction with the robot navigation in rviz, by clicking a location on the map:

https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/

adamconkey, 2022-11-14:
+1 for `moveit_msgs`, in particular the `DisplayRobotState` and `DisplayTrajectory` are incredibly useful for previewing trajectories and visualizing key waypoints in a the trajectory of a robot arm. Is it recommended to create a separate issue for more specific feature requests?

jtbandes, 2022-11-14:
Yes I suppose we could stand to have a separate ticket for the moveit stuff. Is there a reference implementation with screenshots demonstrating how the types are meant to be visualized?

adamconkey, 2022-11-14:
It looks like the visualization code is in [rviz_plugin_render_tools](https://github.com/ros-planning/moveit2/tree/132ad2853960c4cb9da47bdd49dab8ebad2dd723/moveit_ros/visualization/rviz_plugin_render_tools) and in particular there is [trajectory_visualization.cpp](https://github.com/ros-planning/moveit2/blob/132ad2853960c4cb9da47bdd49dab8ebad2dd723/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp). There are some examples in the [videos/images of the MoveIt tutorials](https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) for the trajectory visualization. It looks similar to the URDF visualization in that you're showing the robot with its meshes in the scene, but it's decoupled from a published robot state so that you can e.g. preview a trajectory before executing it, or visualize multiple different plans at once. They also support coloring robot links selectively, which is useful e.g. color links that will collide with an object differently than those that are free of collision.