foxglove/studio #1923: Expose datatype details in extension API

defunctzombie, 2021-10-01:
User Story
> As an extension author, I want to know the "shape" of the data on each topic. This might include which fields are in a message on that topic, are the fields arrays? primitive types? This information lets me build a panel that has topic autocomplete functionality similar to the builtin message path input for panels like Raw Message.

defunctzombie, 2022-04-08:
For additional context on what I see as the path forward to being comfortable exposing the "schemas" to the api in a way we can support.

https://github.com/foxglove/studio/pull/3091#issuecomment-1090598391

banisadr, 2022-12-01:
I think this is no longer relevant unless I'm misunderstanding the description?

defunctzombie, 2022-12-01:
I'm gonna close this out until we have more use cases. The one mentioned here for message path we've exposed to all panels via the _settings_ interface so panels don't need to build that themselves.

closed 2022-12-01 by defunctzombie