foxglove/studio #1925: unfriendly error on web build for unsupported browsers

wkalt, 2021-10-01:
Attached screenshot of studio.foxglove.dev in firefox.
[image]

amacneil, 2021-10-01:
Duplicating some thoughts from Slack:

I think a more user friendly "crashed panel" UI would go a long way here (similar to the "crashed tab" UI in Chrome). We should hide the stacktrace behind a link, and display a sad robot or something instead.

cc @2metres for thoughts

jtbandes, 2021-10-04:
Related: https://github.com/foxglove/studio/issues/1758

jtbandes, 2021-10-05:
Pointer to relevant code: 

https://github.com/foxglove/studio/blob/4fa507eb92d3e68b8b814e8ac1c221c8a5b90c8b/web/src/VersionBanner.tsx#L39-L105

https://github.com/foxglove/studio/blob/586978d02118da940864e9bba87a8a31a298f590/packages/studio-base/src/components/ErrorBoundary.tsx#L41-L56

closed 2022-05-05 by github-actions[bot]