foxglove/studio #1957: Blank error message for oversized ROS2 bags

jhurliman, 2021-10-11:
**Description**

We currently only support ROS2 bags that will fit in the sql.js wasm heap space. When an oversized bag is opened in studio, a very generic error dialog is shown with no error description.

**Expected Behavior**

ROS2 bag loading should have a broad try/catch with a helpful error message that warns users about the lack of support for large bags. Alternatively, if we can dial in an exact upper bound that holds across users and web+desktop we can check the filesize before loading and error out early.

defunctzombie, 2021-12-07:
The first challenge is finding a ros2 bag that is 4gb 😭

jtbandes, 2021-12-08:
Here is example code for how to write a ros2 bag in python: https://github.com/ros2/rosbag2/blob/master/rosbag2_py/test/test_sequential_writer.py#L60

defunctzombie, 2021-12-12:
This line throws when reading a large file:
https://github.com/foxglove/rosbag2-web/blob/main/src/SqliteSqljs.ts#L53

The error thrown has this generic message:
> Error initializing player: The requested file could not be read, typically due to permission problems that have occurred after a reference to a file was acquired.

The name of the error is also a generic `NotReadableError` with code `0`.

The Blob size limit is different on different browsers (https://github.com/eligrey/FileSaver.js/#supported-browsers). The options I see available:

1. Hardcode cap on the size limit and corresponding error message (implemented in rosbag2-web? studio?)
2. Automatic code detection of maximum blob size by making blobs of various sizes...
3. Handle `NotReadableError` in the player and provide our own error message with more ways it could happen (i.e. add note about file size maybe being too large).

defunctzombie, 2021-12-13:
I'm gonna try using a patch for sql.js (https://github.com/foxglove/sql.js/pull/1) which will let us read directly from File instances. This will require moving the Ros2 data provider reading into a worker.

defunctzombie, 2021-12-16:
I was able to patch sql.js to allow reading File instances without loading the file completely into memory. This resulted in some patches to our rosbag2 packages:

https://github.com/foxglove/rosbag2/pull/5 https://github.com/foxglove/rosbag2-web/pull/10

commit 800f9208 (2021-12-17): Add WorkerRosbag2DataProvider

The WorkerRosbag2DataProvider reads rosbag2 .db3 files in a web worker. This leverlages @foxglove/sql.js capabilities to read arbitrary size files without loading them into memory. This capability is only present in workers.

Fixes: #1957
PR #2423 (2021-12-17): Add WorkerRosbag2DataProvider

**User-Facing Changes**
Users are able to load and visualize rosbag2 .db3 files >2GB

**Description**
The WorkerRosbag2DataProvider reads rosbag2 .db3 files in a web worker. This leverlages @foxglove/sql.js capabilities to read arbitrary size files without loading them into memory. This capability is only present in workers.

Fixes: #1957
commit ad7b3538 (2021-12-17): Add WorkerRosbag2DataProvider (#2423)

The WorkerRosbag2DataProvider reads rosbag2 .db3 files in a web worker. This leverlages @foxglove/sql.js capabilities to read arbitrary size files without loading them into memory. This capability is only present in workers.

Fixes: #1957
closed 2021-12-17 by defunctzombie