foxglove/studio #1964: Future: Support preloading (requires larger refactor to allow players other than RandomAccessPlayer to vend preloaded messages)

jtbandes, 2021-10-11:


amacneil, 2022-02-26:
This is working much better, but still not 100%. When loading plots, it seems to preload part of the data, but doesn't load prior to the current seek position, and doesn't preload all the way to the end. Example:

https://user-images.githubusercontent.com/637671/155829545-7e9b8f03-ca13-411b-a22e-4026883b2377.mov

You should be able to repro with [this example](https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_mHH1Cp4gPybCPR8y&time=2018-07-24T03%3A28%3A47.647Z&ds.start=2018-07-24T03%3A28%3A47.647Z&ds.end=2018-07-24T03%3A29%3A06.797Z).

foxymiles, 2022-02-28:
@amacneil There was a bug around preloading segments I have a fix for. Also, currently we preload up to 15 seconds of data. Should I increase that? If so to what?

amacneil, 2022-03-01:
Let's discuss with @defunctzombie.

I don't have data or intuition to say whether increasing preloading from 15 seconds is a good idea. But I think that at some point we will need to bite the bullet and give panels (e.g. plot, map, state transitions) an official way to request complete preloading of certain topics outside of the regular caching system.

defunctzombie, 2022-05-15:
Closing this. The experimental console player does preloading. There's improvements that need to be made there but preloading itself is implemented. I don't want to cut over to the experimental player until I do some better buffering.

PR #2908 (2022-02-25): Expose block cache from foxglove DP player.
**User-Facing Changes**
This will improve preloading behavior for the Foxglove Data Platform player.

**Description**
`FoxgloveDataPlatformPlayer` uses its own memory cache implementation and this cache doesn't expose cached messages in the blocks format that some panels expect. This patch just exposes the cached messages in blocks as well as the format used internally by `FoxgloveDataPlatformPlayer`.

Fixes #1964
closed 2022-05-15 by defunctzombie