foxglove/studio #1988: Extensions API: add ability to read and write parameters

jessicaaustin, 2021-10-18:
The current Extensions API does not provide access to parameters: you cannot read parameters, or set or delete parameters. This capability is available via roslibjs and is used for the Parameters Panel in Studio. 

At least for our use case, reading parameters is much higher priority than updating or deleting them.

jtbandes, 2021-12-10:
Should this issue stay open since it was about both reading & writing parameters? The linked PR only implements writing.

jessicaaustin, 2021-12-10:
Personally I think this should stay open until reading parameters is implemented. @jtbandes

foxymiles, 2021-12-11:
@jessicaaustin Is the parameter access implemented in https://github.com/foxglove/studio/pull/2154 not working for you?

foxymiles, 2021-12-11:
@jtbandes @jessicaaustin Sorry if this wasn't clear but read access to parameters was implemented in #2154 by adding parameters to the `RenderState`. I had to delay implementing write access to parameters until this separate PR because there was an underlying issue in our ROS1 implementation I had to fix. But you should be able to both read and write params via the extension panel API now.

amacneil, 2021-12-13:
Summary: 
- Reading params was added in https://github.com/foxglove/studio/pull/2154
- Writing params was added in https://github.com/foxglove/studio/pull/2316

Both should be working in the latest release (v0.24.0). Please add comments or open a new issue if anything is not working.

PR #2316 (2021-12-09): Add setParameter method to extension panel API.
**User-Facing Changes**
This adds a new `setParameter` method to the extension panel API.

**Description**
This adds a new `setParameter` method to the extension panel API. It has a simple method signature:

```
    setParameter: (name: string, value: ParameterValue) => void;
```

**Note**: for ROS parameter names must begin with a `/` character.

Fixes #1988
closed 2021-12-13 by amacneil