foxglove/studio #1998: Change default font size to 13px

amacneil, 2021-10-20:
**Description**
Studio currently uses a huge mix of base font sizes, from 12px for much of the original webviz design, to 14px for much of the new fluent stuff.

We should standardize on 13px as our base font size for majority of app text. Besides being a middle ground, it's also what vscode uses for most of its interface (besides the code editor).

2metres, 2022-01-24:
This will likely come with the switch over to Material-UI

[image]

defunctzombie, 2022-09-06:
@amacneil @2metres is this still a thing or has our font size situation improved?

amacneil, 2022-09-06:
This is sufficiently done, font sizes are much more consistent now.

closed 2022-09-06 by amacneil