foxglove/studio #2049: Native ROS/ROS2 on web version

bertaveira, 2021-10-29:
It would be amazing to have the functionality of connecting to a live native ROS source while using foxglove on the web version.

We want to use a server docker image with foxglove that is in the same network as the ROS2 system and then we can connect to it via the browser and view live data. This is much harder with the desktop app for our application where we don't want the end users to have access to the actual ROS network just the page itself that's why we use the web version.

I see that you have disabled this feature on the web version since it didn't work in the past but it would be amazing to have ti working instead. Right now this is what it says when we try to connect:

[image]

jtbandes, 2021-10-30:
Hi, thanks for trying out Foxglove Studio!

Native ROS 1/ROS 2 connections in the browser are not possible due to technical limitations of the browser environment. The browser does not allow web apps to make arbitrary TCP/UDP connections, only HTTP/WebSockets/WebRTC.

The recommended way to connect to ROS systems from the web app is to run the Rosbridge WebSocket server on your robot. Read more here: https://foxglove.dev/docs/connection/data-sources We are committed to supporting rosbridge connections so please let us know if you encounter any issues!

closed 2021-10-30 by jtbandes

jtbandes, 2021-11-01:
Adding some more explanatory text in https://github.com/foxglove/studio/pull/2056.