foxglove/studio #2057: Issues encountered when publishing via extensions api (Teleop panel)

jessicaaustin, 2021-11-01:
Background: I'm working on an extension with a panel that publishes messages on a topic. As I was working on that, I came across some issues with how Studio publishes on a topic using the Extensions API. Since I was able to reproduce these issues using the Teleop panel, which also uses that API, the issues below are all using that panel. Please let me know if there is any more info needed. Thanks!

Issues:
* (1) Teleop publisher doesn't re-connect to subscribers on Studio restart
* (2) Teleop publisher doesn't connect to subscribers for `rostopic echo` or Raw Messages panel

Software: Using ROS1 melodic and Studio v0.20.1, on ubuntu 18.04 x86_64

Steps to reproduce (1):

1. Start a new `roscore`
    * Confirm no topics (except rosout)
1. Start up a simple node (listener.py) that listens on `/cmd_vel` and prints messages on that topic:
    ```
    #!/usr/bin/env python3
    
    import rospy
    from geometry_msgs.msg import Twist
    
    def callback(data):
        rospy.loginfo(data)
    
    
    if __name__ == '__main__':
        rospy.init_node('cmd_vel_listener', anonymous=True)
        rospy.Subscriber("cmd_vel", Twist, callback, queue_size=None)
        rospy.spin()
    ```
1. Open Studio
1. Add a teleop panel and configure it to publish to `/cmd_vel`
1. Test sending a message
    * :heavy_check_mark: Confirm it shows up in the node output
    * :heavy_check_mark: `rostopic info /cmd_vel` shows `/foxglovestudio` as a publisher
1. Restart studio
1. Try to send a message using the same teleop panel
    * Expected: message sent, and shows up in the node output
    * Actual:
        * :x:  no message sent
        * :x: `rostopic info /cmd_vel` shows no publishers
        * :x:  studio shows a Problem: `Unable to publish to "/cmd_vel"`. and I don't see anything in Dev Console
    * Workaround:
      * Option 1: remove and re-add the panel
      * Option 2: edit panel settings, remove /cmd_vel as the topic, and re-add it

Steps to reproduce (2):

1. Start a new `roscore` and the listener.py node
2. Open Studio, add a teleop panel and configure it to publish to `/cmd_vel`
3. Add a Raw Messages panel and configure it to listen to `/cmd_vel`
4. Start a `rostopic echo /cmd_vel` process in a terminal
5. Test sending a message using the Teleop panel (see screenshot 885.png below)
    * :heavy_check_mark:  It shows up in the listener.py script
    * :x: It does NOT show up in the Raw Messages panel
        * Warning message in console: `[ROS1] failed to write 0 bytes to tcpros://127.0.0.1:45298: DataCloneError: Failed to execute 'postMessage' on 'MessagePort': ArrayBuffer at index 0 is already detached.`
    * :x: It does NOT show up in `rostopic echo /cmd_vel`
        * Same warning message as above
    * Workaround: none that I could find. I tried starting up the `rostopic echo` script at different times (before adding the teleop panel and after), and it didn't seem to matter, I can never get messages to show up in either of those places. Same thing with the Raw Message panel.
6. Bonus bug!
    * At this point, add a Publish panel and choose `/cmd_vel`
    * After you type `/cmd_vel`, an unhandled error crashes the panel (see screenshot 884.png)

[image]

[image]

defunctzombie, 2021-11-03:
I researched this a bit more and I think it is a bug in our ROS1 (native) connection. I tried the same approach with the ROS Bridge (websocket) connection and was able to open/close the application and still be able to publish to /cmd_vel and have the listener print the received messages.

jessicaaustin, 2021-11-04:
Confirmed I can get things to work when using rosbridge, so that will be my workaround for now. Thank you!

amacneil, 2021-11-09:
Can we split out the second bug into a separate ticket?

defunctzombie, 2021-11-18:
@jhurliman can you take a look at this. I believe it is ROS1 related and might be impacting other users that try to use the publish panel.

defunctzombie, 2021-12-03:
@jhurliman any updates on this?

PR #2582 (2022-01-05): Set publishers before trying to get topics in ros1 player.
**User-Facing Changes**
This fixes an issue with publishing and subscribing to topics on startup.

**Description**
#2574 improved our handling of `advertise` calls made when the ROS1 player was still starting up by queing them and retrying them once the player was ready. This builds on that fix to wait to request topics until that second attempt to advertise topics is made so that any freshly advertised topics are available to subscribers.

Fixes #2057
closed 2022-01-05 by foxymiles