foxglove/studio #2064: AVC / H.264 video support

amacneil, 2021-11-02:
Currently Foxglove Studio supports rendering image frames using the [sensor_msgs/Image](http://docs.ros.org/en/api/sensor_msgs/html/msg/Image.html) and [sensor_msgs/CompressedImage](http://docs.ros.org/en/api/sensor_msgs/html/msg/CompressedImage.html). The latter supports PNG or JPEG compression.

Several people have asked for the ability to render video data, most commonly in h264 format. Often, this data is recorded as a separate file alongside a ROS bag, but in theory the NAL units could be wrapped within a ROS or Protobuf message.

Several methods to implement this have been discussed, from using native HTML video elements, to decoding the frames in WASM.

Related tickets: https://github.com/foxglove/studio/issues/1069 https://github.com/foxglove/studio/issues/537

Aposhian, 2022-01-05:
This would be very useful. ROS Image topics are horribly inefficent, since they are raw image frames without intra-frame or frame-to-frame compression (like H.264). Specifically, being able to subscribe to RTSP streams would be amazing.

jtbandes, 2022-01-05:
@Aposhian How do you imagine those streams would be synced with other message data coming from the main data source?

Aposhian, 2022-01-05:
I suppose what I was envisioning was primarily supporting the live use case, where you are streaming from a dedicated video source that is assumed to be synced with your primary data source. However, absolute timing information can be obtained from a container protocol like RTSP. While I may be incorrect on this, it seems that RTCP messages can contain NTP timestamps, and RTP packets contain relative timestamps, so that by combining the two, you can obtain an absolute timestamp for each RTP packet. Then, if your RTSP source is synced with the same NTP server as your primary data source, it should be synced. I've primarily used RTSP in my applications, so I'm unsure of what container formats with H.264 data other people would want to visualize in foxglove.

amacneil, 2022-06-16:
Related: We should support visualizing h264 directly in MCAP (https://github.com/foxglove/mcap/issues/427)

lishulincug, 2022-06-28:
https://github.com/foxglove/studio/issues/3739#issue-1286687657

lishulincug, 2022-06-28:
how to translat rosbag to RTSP 

> I suppose what I was envisioning was primarily supporting the live use case, where you are streaming from a dedicated video source that is assumed to be synced with your primary data source. [...]

How to set the rosbag's image into a RTSP

lishulincug, 2022-06-28:
> I suppose what I was envisioning was primarily supporting the live use case, where you are streaming from a dedicated video source that is assumed to be synced with your primary data source. [...]

How to set the rosbag's image into a RTSP

Aposhian, 2022-07-11:
Not sure if this is the correct way to go about it, but it might also be good to take note of https://github.com/clydemcqueen/h264_image_transport