foxglove/studio #2068: Layout change detection is non-deterministic

bclinkinbeard, 2021-11-02:
Given a saved layout that includes a Plot panel with the legend open, close the legend and then reopen it. The layouts panel reports a change to the layout, even though the state matches the initial value.

This is reproducible on https://studio.foxglove.dev/

**Steps To Reproduce**
See above.

**Expected Behavior**
Layout state is matched to the initial state, and there are no pending changes.

**Actual Behavior**
Layout panel still reports pending changes, even though the state is the same.

asyncanup, 2021-11-12:
@defunctzombie
1. Could you clarify why this is a feature, not a bug?
2. Do you have an ETA for implementing this change?

defunctzombie, 2021-11-13:
@asyncanup Every panel has a configuration (or state) object that it saves as part of the layout. The Plot panel considers the legend toggle state as part of this persisted configuration so when a user toggles the legend the plot panel saves the new state. Studio considers every state save as an indicator the layout is "changed".

I marked this as a feature since the behavior currently works as designed (panel says save this new state for me so the layout is considered "changed") and the feature is to enhance that change detection logic to either doing more detection around whether the state is different from the saved state or possibly having some features like the state of the legend not be relevant or part of the saved layout state.

Whether it's a bug vs feature is semantics. Certainly it's undesirable? annoying? behavior from your standpoint since visually the layout is unchanged but I'd also argue nothing is broken or preventing use of the app.

There is no ETA for implementing this change.

defunctzombie, 2022-03-10:
@jtbandes See: https://github.com/foxglove/nvidia-issue-tracker/issues/42#issuecomment-1063525384

PR #3009 (2022-03-10): LayoutManager: remove working copy when edits result in same data
**User-Facing Changes**
An edited layout will no longer appear modified in the sidebar if the edits result in the same configuration data. 

**Description**
Fixes #2068
cc @bclinkinbeard 

We simply remove the `working` state if the edits result in config data that is `isEqual` with the baseline data.
commit 9c3fb80e (2022-03-10): LayoutManager: remove working copy when edits result in same data (#3009)

**User-Facing Changes**
An edited layout will no longer appear modified in the sidebar if the edits result in the same configuration data. 

**Description**
Fixes #2068
cc @bclinkinbeard 

We simply remove the `working` state if the edits result in config data that is `isEqual` with the baseline data.
closed 2022-03-10 by jtbandes