foxglove/studio #2072: Non-standard `utf8` encoding used instead of `utf-8`

Edholm, 2021-11-03:
**Description**

Foxglove Studio uses non-standard `utf8` encoding instead of `utf-8`.

- Version: v0.21.0
- OS: Linux

**Steps To Reproduce**
Connect to a node that has been created by `[goroslib](https://github.com/aler9/goroslib)`, receive 400 Bad Request back.

**Expected Behavior**
xmlrpc calls work with `encoding=utf-8` instead of `encoding=utf8` 

**Actual Behavior**
When Foxglove Studio connects to the node using xmlrpc it sends an `utf8` encoding string instead of the expected `utf-8`. This causes the node to return 400 Bad Request since it only supports `utf-8`.

I don't know if this issue should be opened in the foxglove/xmlrpc repo, but the behavior is visible here so I opened the issue here.

I opened a [PR](https://github.com/aler9/goroslib/pull/91) to fix the issue in `goroslib`, but the author pointed out that it might be better to solve the issue here instead.

jtbandes, 2021-11-04:
Thank you for the report!

We are open to making this change, but I'd like to see a definitive source for the claim that `utf-8` is the most correct/standard value. Or should we use `UTF-8`?

Would you be willing to open a PR against foxglove/xmlrpc for the change?

Edholm, 2021-11-05:
> Thank you for the report!
> 
> We are open to making this change, but I'd like to see a definitive source for the claim that `utf-8` is the most correct/standard value. Or should we use `UTF-8`?
> 
> Would you be willing to open a PR against foxglove/xmlrpc for the change?

Yes, `UTF-8` is probably even more correct since that is what the [RFC](https://datatracker.ietf.org/doc/html/rfc3629) uses it seems. The problem for me was the lack of `-`. 

I can open a PR in the repo :+1:

closed 2021-11-05 by jhurliman