foxglove/studio #2085: Unable to log in to dev build of web app

jtbandes, 2021-11-05:
**Steps To Reproduce**
Run `yarn web:serve`. Visit localhost:8080 and try to log in.

**Expected Behavior**
Login flow is successful

**Actual Behavior**
I'm redirected to the login page, then back to the app, but nothing happens

foxymiles, 2021-11-19:
Looking into this. This is going to require some changes to our auth flow it looks like.

cc @bryfox

foxymiles, 2021-12-08:
To make this a little easier in the meantime while we work on a better fix I've added two new package.json scripts:

```yarn web:serve:dev```

and 

```yarn web:serve:local```

defunctzombie, 2022-01-06:
https://github.com/foxglove/studio/issues/2086#issuecomment-1006768155

defunctzombie, 2022-01-06:
I've removed the bug label from this issue. It impacts local development of the web build only. I added some comments on a related ticket. We can consider adding some different flow when in `NODE_ENV===development` or at least a notice.

defunctzombie, 2022-07-06:
We use device auth flow in the dev build now so you can login.

closed 2022-07-06 by defunctzombie