foxglove/studio #2086: Login failure should display an error to the user

jtbandes, 2021-11-05:
**Steps To Reproduce**
Follow steps from #2085

**Expected Behavior**
When login fails, an error should be visible to the user

**Actual Behavior**
The failure is silent

amacneil, 2021-11-18:
Is this for web or desktop or both?

defunctzombie, 2022-01-06:
I'm gonna take a look at this issue and see what the impact is (dev or production) and report back.

defunctzombie, 2022-01-06:
This issue is specific to the web app in development mode.

The flow that occurs for the web app is to redirect to the console (production) signin page with a `returnTo` url argument back to localhost.

Console is already signed in (and cookies are set) and redirects back to the returnTo url (localhost). Since localhost is not able to access a production (foxglove.dev) cookie the user remains signed out.

One improvement we can make is to detect unsupported returnTo urls on console (oauth does this) and avoid doing the returnTo. Technically we aren't doing what oauth does tho since our returnTo arg to the console url is a redirect and does not contain any tokens in the url. That is why we don't verify the returnTo url.

defunctzombie, 2022-01-06:
Another idea is to open the signin url in a new tab and poll in the existing tab (studio) waiting for the user to signin. Overall I think the experience we have today for production works smoothly but the dev experience does not for the web build and that is a devx enhancement.

I am closing this issue in favor of #2086 and track that as a developer experience enhancement.

closed 2022-01-06 by defunctzombie