foxglove/studio #2091: Adding first panel to layout takes you to panel settings, additional panels do not

amacneil, 2021-11-05:
**Description**
- Version: v0.20.1
- OS: macOS, web

**Steps To Reproduce**
1. create a new layout
2. add a panel to it. observe: left sidebar changes to "panel settings"
3. go back to "add panel" sidebar and add another panel. observe: left sidebar stays in "add panel" mode

**Expected Behavior**
Consistency between these two actions.

I would prefer we never take you to the panel settings page after adding a panel. More often than not when building a layout I want to add multiple panels before configuring any settings, and it is jarring having the sidebar change after adding a panel.

jtbandes, 2021-11-07:
Seems to be a difference between drag/drop and just clicking an item in the list. Clicking works as expected.

jtbandes, 2021-11-07:
> I would prefer we never take you to the panel settings page after adding a panel. More often than not when building a layout I want to add multiple panels before configuring any settings, and it is jarring having the sidebar change after adding a panel.

The point of automatically opening settings was discoverability. For new users, it's probably not obvious how to configure panels so we wanted to start teaching them to use the sidebar. Any ideas for other ways we can improve discoverability?

amacneil, 2021-11-07:
> Seems to be a difference between drag/drop and just clicking an item in the list. Clicking works as expected.

Interesting. Clicking seems to always take you to the panel settings. Dragging only takes you to the panel settings for the first panel; subsequent panels do not take you to panel settings.

> The point of automatically opening settings was discoverability. For new users, it's probably not obvious how to configure panels so we wanted to start teaching them to use the sidebar. Any ideas for other ways we can improve discoverability?

We have a cog icon on every panel, and clicking it shows you the `Panel settings` option. That seems pretty discoverable?

Since many panel options (e.g. selected topic) are displayed directly in the panel, I don't find the panel settings something that is immediately needed while building a layout.

Curious to hear what others think of this behavior. cc @defunctzombie @jhurliman @2metres @esthersweon.

At the very least, we should fix the inconsistency, so that adding a panel either always opens the panel settings or never opens the panel settings.

amacneil, 2021-11-11:
per slack conversation - let's proceed with never automatically opening sidebar settings for now. we can revisit in the future once we are happier with how panel configuration works.

closed 2021-11-13 by jtbandes