foxglove/studio #2097: Worldview backgroundColor in Studio does not do anything

jhurliman, 2021-11-06:
See https://github.com/foxglove/studio/pull/2096

Even when no scene content is drawn, setting the backgroundColor to solid green still produces a black 3D scene.

jtbandes, 2021-11-07:
Is it possible you're using the wrong units (0–1 vs 0–255)?

jhurliman, 2021-11-08:
I tried both 1 and 255 for the red value, same result of a black 3D view.

jtbandes, 2021-11-08:
I added `backgroundColor={[1, 0, 0, 1]}` in my local branch and it works for me. I don't think there is a Studio bug here. However, it does seem like maybe regl-worldview doesn't support dynamically changing the background color?

[image]

closed 2021-11-08 by jtbandes