foxglove/studio #2098: Studio doesn't render mesh Resoure Markers with the correct opacity

alluring-mushroom, 2021-11-07:
**Description**
When broadcasting ros2 Markers, with the opacity less than 1.0, studio renders the mesh correctly, except it has no transparency.

- Version: Foxglove Studio 0.21.0
- OS: Ubuntu 20.04

**Steps To Reproduce**

- Setup a python node to broadcast a Marker
[image]

- the url here is http://localhost:8000/models/discharge_flask/discharge_flask.glb, and is being served by the builtin python http server for now
- I am running in ros2 galactic, using the ros2 web-bridge to get the data to foxglove 

**Expected Behavior**
It should look like this (this is my hacky implementation of meshmarkers I made before the awesome foxglove team worked on it
[image]

**Actual Behavior**
No transparency is rendered:
[image]
Instead

alluring-mushroom, 2021-11-08:
Sorry, I forgot to submit the bag. Here it i
[rosbag2_2021_11_08-10_01_13.zip](https://github.com/foxglove/studio/files/7493411/rosbag2_2021_11_08-10_01_13.zip)
s

jhurliman, 2021-11-10:
@alluring-mushroom can you send http://localhost:8000/models/discharge_flask/discharge_flask.glb as well?

jhurliman, 2021-11-10:
@alluring-mushroom never mind, I was able to confirm opacity works in my PR without it. Fix is landing shortly.

closed 2021-11-10 by jhurliman