foxglove/studio #2099: Rosout panel settings disappears every time a new message appears in the panel

munzz11, 2021-11-08:
**Description**

- Version: 0.21.0
- OS: Ubuntu 20.04

**Steps To Reproduce**
1. Connect to ROS1 running on a separate machine
2. Add rosout panel to empty layout and start viewing rosout messages
3. Click the settings cog for the panel, every time there is a new message, the panel settings drop down disappears, meaning if you have a lot of log messages coming through, it can become almost impossible to change the panel settings  

**Expected Behavior**
Have the panel settings menu not disappear and be able to interact with it, when messages are coming through 

**Actual Behavior**
[image]

munzz11, 2021-11-08:
Source link of gif since is appears to freeze in github: https://i.ibb.co/CQYqj4G/ezgif-com-gif-maker.gif

foxymiles, 2021-11-09:
This looks like it's a consequence of fluentui's dismiss on scroll behavior:

https://github.com/microsoft/fluentui/issues/18080

I'm investigating some possible solutions.

closed 2021-11-09 by jtbandes