foxglove/studio #2117: Add possibility to use layouts hosted on a robot.

adrienbarral, 2021-11-09:
Hello, as discussed by mail with @amacneil and @defunctzombie , here is a public feature request open for discussion.

I have a use case that is not completely fulfilled by Foxglove, and I would like to propose to you a Pull Request.

In my firm we build ROS1 and ROS2 based robots. We plan to use FoxGlove to propose some maintenance pages for our robots. Most of the time, when users will use these maintenance pages, they will be offline (without any internet connection, and we can't assume that they already had one). I would like to serve foxglove web on a webserver hosted on the robot. And I would like to have some predefined layouts hosted on the robot. These predefined layouts will be designed by the robots development teams. They haven’t  vocation to be modifiable. 

Today I do this with webviz by using the URL parameter :"layout-url", and having some JSON layout hosted on my robot. That works like a charm. I read the shared gdoc document “FoxGlove studio deeplink”, and my use case doesn't seem to be addressed. (https://docs.google.com/document/d/1jULICvQaZbJM8Et4ylFCE7GIFwedq7YkB4NhvnpjW0k/edit#heading=h.ic2oeiqyrxid)

What I plan to do is to provide a backend service (that we will publish on github (or that can be part of the foxglove repo, I don’t care) ) that will serve some layouts. This service will run on our robots. Then, we could create a new class in FoxGlove Studio that implements the IRemoteLayoutStorage Service to connect to this backend. Of course our backend service will be very simple. We don't plan to address any authentication mechanism, and it will not be designed to be hosted elsewhere than in a local area network. So, someone that needs to use a layout stored on the web and shared with teammates will continue to use your existing service. 

Do you think we can work on this feature and propose an implementation ? 

Did you already plan to address my use case by "selling" the backend service that I can host on my robots ?

defunctzombie, 2021-11-11:
> Then, we could create a new class in FoxGlove Studio that implements the IRemoteLayoutStorage Service to connect to this backend

Yep - this sounds right. Depending on where/how you are deploying what sounds like your own copy of Foxglove Studio you might be able to avoid making a separate backend service and use the same webserver that serves up the web version of Studio to also serve up layout files and have your implementation of IRemoteLayoutStorage read those files on startup and keep the remaining operations in-memory.

Yet another option could be to build your layouts into your custom web build of studio that is deployed on the robots.

We don't have immediate plans to address the specific use case where you want to host the layouts on your robot.

amacneil, 2022-05-06:
@adrienbarral would your needs be covered by https://github.com/foxglove/studio/issues/2792? I know that issue is more recent than yours, but it has more active discussion so I think we should merge your ticket with it.

adrienbarral, 2022-05-09:
@amacneil  : Yes you're right. I think we can close my ticket, and follow #2792

closed 2022-05-09 by adrienbarral