foxglove/studio #2133: Swapping out a panel produces panel with no settings

jtbandes, 2021-11-10:
**Steps To Reproduce**
- Add a 3D panel. Click ⚙️ > Panel Settings and observe defaults are present (2/3 switches are on by default).
- Click ⚙️ > Change panel > 3D. Look at the new panel's settings.

**Expected Behavior**
The swapped-in 3D panel should have the same defaults set.

**Actual Behavior**
The new panel has a config of `{}`, which for some reason does not get merged with the 3D panel's defaultConfig.

If swapping in a new Tab panel, this bug causes the newly added tab's "Pick a panel" to not work.

amacneil, 2021-11-18:
Let's also prevent users from changing their current panel to the same panel type.

defunctzombie, 2021-12-03:
Did this get looked at? I recall us looking at it but wasn't sure if this rolled into any of the various layout PRs that may have fixed this.

I tried reproducing by swapping in a new tab panel and was able to use the "pick a panel" feature of the tab panel and was able to pick a panel.

jtbandes, 2021-12-03:
That particular Tab issue was fixed by https://github.com/foxglove/studio/pull/2132

PR #2296 (2021-12-07): Set panel default config when config is empty
**User-Facing Changes**
When a user swaps a panel for another and then opens the panel settings they will see the correct defaults.

**Description**
Previously the Panel wrapper would set the default config only if the config was undefined. The config is sometimes defined but as an empty object (panel swap). In this instance the default config should also be set so the panel and panel settings receives the default values.
    
Fixes: #2133
commit ed4ac014 (2021-12-07): Set panel default config when config is empty (#2296)

Previously the Panel wrapper would set the default config only if the config was undefined. The config is sometimes defined but as an empty object (panel swap). In this instance the default config should also be set so the panel and panel settings receives the default values.

Fixes: #2133
closed 2021-12-07 by defunctzombie