foxglove/studio #2148: Improve welcome experience

amacneil, 2021-11-11:
Our current welcome layout is pretty overwhelming. We should provide a nicer first-run experience, showing the power of studio and also walking users through some basic things they can do.

defunctzombie, 2021-11-12:
Idea: remove the welcome panel and replace it with a modal that shows on first run. The modal can be one of those carousel UI patterns so the user can scroll through some screens with app features.

defunctzombie, 2021-11-30:
In-app feedback.

defunctzombie, 2021-12-24:
@amacneil `Refreshing page with a local bag file loaded should keep that bag file loaded`

It is a browser limitation that we must ask for user consent before we have access on refresh. If we are ok with that - then the ask here is that we remember that you have a local file open and automatically ask for consent? Similar to how we remember connection based data sources and try to reconnect when they are specific in the url args?

amacneil, 2021-12-24:
Yeah I think that (asking for consent on refresh) would be better than dumping the user back onto the welcome dialog.

defunctzombie, 2022-01-06:
@amacneil I've removed `- [ ] Refreshing page with a local bag file loaded should keep that bag file loaded` to bring down the scope of this issue. We have the recents list which is working well. I don't believe this specific issue is that impactful to new user experience.

amacneil, 2022-01-06:
👍 Maybe it's not part of the new user experience, but seems still worth creating a separate ticket for? Refreshing seems fairly common with our users (if they are encountering unexpected issues/errors). Could we keep some sort of reference (even if opaque identifier) in the URL to facilitate that?

defunctzombie, 2022-01-06:
@amacneil we can definitely make this work - I just wanted to scope it out of this work so we can have a finite set of things we are trying to finish here.

I agree refreshing is common - I certainly do it all the time with development - but I've also found that the recents list has made me 99% happy when I refresh and can click the first item so think the additional win is low for local files right now.

defunctzombie, 2022-01-08:
> * Remove start/end/duration for live connections

For now the end time here is the only place a user can see the current timestamp since we do not show any playback controls. So we should keep these and revisit later.

closed 2022-01-18 by amacneil