foxglove/studio #2151: Unable to native connect to ROS2 foxy

bertaveira, 2021-11-13:
**Description**
Running on the local machine under any domain id I can't get foxglove to detect the running nodes and topics. I don't know if the ROS time is supposed to start from 0 every time I connect again or if it indicates that it is indeed not detecting the ROS instances running the the local machine.

- Version: 0.20.1
- OS: Ubuntu 20.04 (Foxy)

**Steps To Reproduce**

Run any node or play any bag file using ROS foxy. I've tried with both FastDDS (foxy default) and CycloneDDS and none are detected.

**Expected Behavior**
Foxglove would be able to connect to the ROS instance and show the topics being transmitted.

**Actual Behavior**
[image]

jtbandes, 2021-11-13:
Based on the log in the screenshot, this is related to https://github.com/foxglove/studio/issues/2030.

jhurliman, 2021-11-25:
@bertaveira I'm having trouble reproducing this. Can you try the latest version of Studio, and in the Chrome dev tools console enable Verbose logging (using the console gear icon, click the "Default levels" dropdown and select "Verbose") then try to connect to your ROS2 source? Copy the full log and attach it here either in a comment or a .txt attachment.

A few other questions about your setup to help reproduce:

* Did you build ROS 2 from source or install apt packages or some other method?
* Is everything running on the same operating system as Studio or are you using multiple computers, virtual machines, Docker instances, etc?
* Are you switching middleware using `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` or some other method?
* Can you try a simple demo app by running `ros2 run demo_nodes_cpp talker` and see if Studio will connect to that?

bertaveira, 2021-11-25:
@jhurliman Hi. Thanks for asking. I have attached the log file using the simple talker demo.
[foxglove_error.log](https://github.com/foxglove/studio/files/7605202/foxglove_error.log)

Answering your questions:
- I installed using apt (currently on the v0.22.1)
- I am running all in the local machine. Tried both with ros2 bag play and with real nodes running
- I tried both with `rmw_cyclonedds_cpp` and `rmw_fastrtps_cpp` and the resolt is the same
- I have the same results. The attached log file was done

jhurliman, 2021-11-29:
Great, I was able to reproduce the issue and fix it: https://github.com/foxglove/studio/pull/2223

closed 2021-11-29 by jhurliman