foxglove/studio #2156: Tech Debt: Make panel directories, titles, and types consistent with each other

esthersweon, 2021-11-15:
- Should include plan to migrate legacy layouts to use updated panel types

defunctzombie, 2022-05-15:
@esthersweon Is there anything outstanding to do here? I think we've tackled any relevant items as we've iterated on the code.

esthersweon, 2022-05-16:
@defunctzombie - I still see the Log panel with a `panelType` of `"RosOut"`, and the Image panel with a `panelType` and directory name of `ImageView`.

Likely that there are other inconsistencies.

Not an urgent issue, but feels like something we should do to make contributing easier for open source folks.

defunctzombie, 2022-06-08:
@esthersweon We are not planning on changing the panelType since that would break existing layouts. We could create logic for backwards compat for hose but it seems of little value. Probably we can close this ticket out and if there are specific enhancements we need to do that come up in the context of bugs or features we can do them.

closed 2022-09-08 by jtbandes