foxglove/studio #2169: Out of memory when playing back with '--clock'

zebrapants, 2021-11-17:
**Description**
I started playback of a set of bag files with '--clock' parameter set, then started Foxglove Studio with two image views, and after a short while, I got the message "Error: Failed to execute 'postMessage' on 'Worker': Data cannot be cloned, out of memory" in both image views. At this point, the memory consumption of Foxglove Studio has slowly climbed (not jumped) from 0.8 % to around 22 % of my entire system.

It appears the memory consumption also steadily increases with no image views, just not as fast.

The reluctance to release the memory only occurs when I use the '--clock' parameter.

The error log shows:
Error: failed to allocate 9463030 bytes to concatenate 157 chunks
    at Object.e [as concatData] (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:6:93494)
    at h.addData (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:6:79395)
    at bu._handleData (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:6:87751)
    at bu.t.emit (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:253:71633)
    at file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3691:7379
    at MessagePort.x.onmessage (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3691:7625)

- Version: Foxglove Studio 0.22.0
- OS: Ubuntu 20.04
- ROS Noetic
- 32 GB memory

**Steps To Reproduce**

1. Start roscore
2. Start Foxglove Studio (connect to localhost, add panel on image topic)
3. Rosbag play a bunch of bags with '--clock'
4. Watch the memory consumption increase

[image]

**Expected Behavior**
- Show image and not error message
- Stable memory consumption

**Actual Behavior**
- Error message
- Monotonically increasing memory consumption

defunctzombie, 2021-11-17:
@zebrapants To aid with reproducing - does the error happen on the first play through of the bag file or subsequent plays?

zebrapants, 2021-11-18:
@defunctzombie Hi, thanks for picking this up. 
It happens on the first playthrough as well as on subsequent playthroughs. I.e. I can see the memory consumption increasing from the first playthrough, and not released when the bag is finished. Subsequent playthroughs increases the memory consumption further until it eventually runs out of memory. 

I usually play bag files like "rosbag play --clock mybagfile_*.bag" because, as you know, bag files with images tend to get very large, so I split them into multiple files - this is what I meant by a bunch of bags.

Just to be clear, I did try playing back a single bag file, which also resulted in unreleased memory, so it does not appear to be related to the number of bags played back.

PR #2185 (2021-11-20): Clear incorrect messages from out-of-order messages after warning
**User-Facing Changes**
When replaying bag data with the `--clock` flag, Studio would continue to accumulate messages in memory until eventually crashing. This change fixes the memory accumulation behavior so the app does not crash.

**Description**
This change converts warnOnOutOfSyncMessages to a class so we can instantiate a new instance when players change. This resets message order tracking when changing players.
    
The new class fixes a memory issue with keeping all messages over the drift threshold in memory. The timeou [... cut]
commit dbb7b542 (2021-11-20): Clear incorrect messages from out-of-order messages after warning (#2185)

This change converts warnOnOutOfSyncMessages to a class so we can instantiate a new instance when players change. This resets message order tracking when changing players.

The new class fixes a memory issue with keeping all messages over the drift threshold in memory. The timeout which prints the log warning failed to clear the existing incorrect message array _and_ failed to clear the warningTimeout variable. The last bug prevented any new warning timeout from triggering so the array could be cleared.

Fixes: #2169
closed 2021-11-20 by jtbandes