foxglove/studio #2181: Overlay 3D markers on camera images in Image panel

esthersweon, 2021-11-18:


amacneil, 2022-05-17:
We're currently in the process of a ground up rewrite of the 3d panel. That project is a prerequisite for rendering 3D markers into the image panel. So it's definitely on our roadmap, but probably not for a couple months. In the meantime, the workaround is to calculate the [ImageMarker](https://foxglove.dev/docs/studio/panels/image#imageannotation) positions yourself and publish them as a separate topic.

gilduz, 2022-09-14:
Hi all, is there any news about this topic? Including rendering of point cloud on top of 2D images.

amacneil, 2022-09-14:
We just launched the new 3D panel out of beta this week, so it will need some time for that to bake and fix any issues that come up. We do not currently have a planned timeline for rendering 3d objects or point clouds in the image panel.

gilduz, 2022-09-14:
Thanks for the update. I tried new 3D panel and it's great. The ability to
render 3D objects on image panel will be ultimate feature to definitely
switch to foxglove.

adamconkey, 2022-11-12:
This is the main feature that's in rviz that I definitely want in Foxglove.