foxglove/studio #2186: Point cloud decoding failed error on empty Pointcloud

gilduz, 2021-11-19:
**Description**
Pointcloud data with empty datafield generate error "Point cloud decoding failed"

- Version: Foxglove Studio v0.22.1
- OS: Ubuntu 18.04

**Steps To Reproduce**

Create a Pointcloud2 msg with the following content:

```
header: 
  seq: 1635729
  stamp: 
    secs: 1637311582
    nsecs: 506327942
  frame_id: "world"
height: 1
width: 0
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
is_bigendian: False
point_step: 16
row_step: 0
data: []
is_dense: False
```
Got the following error on every received message

[image]

**Expected Behavior**
Reproduce Rviz behaviour: nothing displayed since pointcloud is empty but without errors

**Actual Behavior**
"Point cloud decoding failed" error displayed on every message received

defunctzombie, 2021-11-19:
Thanks for reporting!

Along with rendering nothing since there is no data; would it be useful to indicate any sort of notice next to the topic (in the topic picker) that the point cloud data is empty when this does happen?

gilduz, 2021-11-19:
Hello!
I think that this could be useful but not mandatory, in RVIZ is possible to detect this behaviour since the message counter keeps increasing on every msg received even if contains no data.

Thanks!

PR #2390 (2021-12-15): Improve handling of point cloud render errors.
**User-Facing Changes**
This improves point cloud error handling.

**Description**
The solution is to inspect point cloud messages in `SceneBuilder`. If they are in a big-endian format, which we don't support yet, or have no point data, we register an error on the topic in the topic tree.

Fixes #2186 
Fixes #2225
closed 2021-12-15 by foxymiles