foxglove/studio #2199: URDF viz disappears when toggling between tabs

jhurliman, 2021-11-22:
**Description**

Create a layout with two or more tabs, at least one containing a 3D scene. Set the 3D Model in that scene to a valid URDF URL so the robot model appears. Switch tabs, then switch back. The model and any transform visualizations loaded from the URDF will have disappeared until you change the camera coordinate frame.

- Version: `db448473a975cdd3da17d829a8e86d8de532368c `
- OS: All

defunctzombie, 2022-01-06:
@jhurliman sending your way - I think you added the URDF feature to the 3d panel so hoping you have some context on tackling this.

defunctzombie, 2022-01-06:
Spoke with @jhurliman - gonna grab this since it might be a strange react-ism we need to apply to resolve this.

commit f27b9bd2 (2022-01-07): Fix transform loading from URDF robot models in 3d panel

URDF models may contain joints. These joints represent transforms between different frames of the robot. 3D panel is able to add this frame information to the transform tree.

This change fixes update behavior in the 3d panel when loading URDF models with joints. Previously, the UrdfBuilder would add the joint transforms directly to the transform tree. This approach did not inform the react loop to run or for the transform tree to invalidate itself and create a new instance (as required by hook dependencies).

Instead, UrdfBuilder is no [... cut]
PR #2602 (2022-01-07): Fix transform loading from URDF robot models in 3d panel
**User-Facing Changes**
The 3d scene updates with newly added transforms when setting the robot model.

**Description**

URDF models may contain joints. These joints represent transforms between different frames of the robot. 3D panel is able to add this frame information to the transform tree.

This change fixes update behavior in the 3d panel when loading URDF models with joints. Previously, the UrdfBuilder would add the joint transforms directly to the transform tree. This approach did not inform the react loop to run or for the trans [... cut]
commit b213ed8d (2022-01-07): Fix transform loading from URDF robot models in 3d panel (#2602)

URDF models may contain joints. These joints represent transforms between different frames of the robot. 3D panel is able to add this frame information to the transform tree.

This change fixes update behavior in the 3d panel when loading URDF models with joints. Previously, the UrdfBuilder would add the joint transforms directly to the transform tree. This approach did not inform the react loop to run or for the transform tree to invalidate itself and create a new instance (as required by hook dependencies).

Instead, UrdfBuild [... cut]
closed 2022-01-07 by defunctzombie