foxglove/studio #2203: URDF visualization fails for duckiebot

jhurliman, 2021-11-22:
One or more bugs in xacro-parser prevents the URDF in duckiebot_description from being loaded. This appears to be a separate issue from https://github.com/foxglove/studio/issues/2202, hence the separate ticket.

jhurliman, 2021-12-17:
Should be fixed in https://github.com/gkjohnson/xacro-parser/pull/76

closed 2021-12-21 by jhurliman

defunctzombie, 2022-01-03:
@jhurliman Which urdf file did you use to test this?

jhurliman, 2022-01-03:
https://github.com/duckietown/duckietown-sim-server/blob/master/duckiebot_description/urdf/duckiebot.xacro

defunctzombie, 2022-01-03:
@jhurliman Did you have to also install the package locally on your computer cause it references a package:// url? I tried opening in our web app and it didn't work.

jhurliman, 2022-01-06:
Yes, I've only tested non-trivial URDF loading with package:// URLs.