foxglove/studio #2204: UI Regression: Grouping panels as tabs

esthersweon, 2021-11-23:
**Description**
Regression in UI for grouping panels as tabs

Before:
[image]

After:
[image]

- Version: 0.22.1
- OS: macOS Monterey

**Steps To Reproduce**

- `Cmd` + click multiple panels to select
- Hover over any selected panel

**Expected Behavior**
- See consistently colored buttons with spacing between them

**Actual Behavior**
- See inconsistently colored buttons with no spacing between them

jtbandes, 2021-11-23:
I think I fixed this in #2194, can you verify on main?

2metres, 2021-11-23:
Still seeing in v22.1

2metres, 2021-11-23:
Appears fixed on `main`

closed 2021-11-23 by 2metres