foxglove/studio #2205: "Change panel" menu jumps down screen when typing in a search term

esthersweon, 2021-11-23:
**Description**
- "Change panel" menu jumps down screen when typing in a search term

- Version: 0.22.1
- OS: macOS Monterey

**Steps To Reproduce**

- Open any panel's menu
- Click into "Choose panel"
- Start typing in a search term
  [image]
  [image]

**Expected Behavior**
- Submenu stays connected to the "Choose panel" menu option

**Actual Behavior**
- Submenu jumps down the screen as the submenu gets shorter

jtbandes, 2021-11-24:
I think this might be a recent regression...I'm pretty sure I used this menu within the last week and didn't see this behavior, but I can reproduce on studio.foxglove.dev right now

jtbandes, 2021-12-06:
Doesn't seem to be fully fixed. I think it depends on the initial screen size and whether the full menu content fits on screen without scrolling.

PR #2234 (2021-12-01): Workaround for Change Panel menu positioning
**User-Facing Changes**
Fixed a visual bug with the alignment of the "Change panel" menu.

**Description**
Fixes #2205 by making the menu full height even when it is filtered.

The underlying problem is that fluentui's Callout doesn't reposition itself when the content size changes (https://github.com/microsoft/fluentui/issues/18839). This appears to have been addressed in https://github.com/microsoft/fluentui/pull/18887 but later reverted and the original issue is still unresolved.

[image]
commit 9cd1383d (2021-12-01): Workaround for Change Panel menu positioning (#2234)

**User-Facing Changes**
Fixed a visual bug with the alignment of the "Change panel" menu.

**Description**
Fixes #2205 by making the menu full height even when it is filtered.

The underlying problem is that fluentui's Callout doesn't reposition itself when the content size changes (https://github.com/microsoft/fluentui/issues/18839). This appears to have been addressed in https://github.com/microsoft/fluentui/pull/18887 but later reverted and the original issue is still unresolved.
PR #2290 (2021-12-07): Use fixed-height context menu for ChangePanel

**User-Facing Changes**
Fixed a visual bug with the alignment of the "Change panel" menu.

**Description**
The original fix (https://github.com/foxglove/studio/commit/9cd1383dda4ee1f6c71f240ec87cfdf82c0cc5be) did not work in all cases. This change forces the subcontext menu height to a fixed height so it stays anchored to its top edge as the panel list changes when a user searches.

Fixes: #2205
commit 174db118 (2021-12-07): Use fixed-height context menu for ChangePanel (#2290)

The original fix (https://github.com/foxglove/studio/commit/9cd1383dda4ee1f6c71f240ec87cfdf82c0cc5be) did not work in all cases. This change forces the subcontext menu height to a fixed height so it stays anchored to its top edge as the panel list changes when a user searches.

Fixes: #2205
closed 2021-12-07 by defunctzombie