foxglove/studio #2210: Remove requirement for node playground nodes to have `/studio_node/` prefix

amacneil, 2021-11-23:
Currently node playground forces you to have a `/studio_node/` prefix. This behavior is inherited from webviz, but seems unnecessarily restrictive. For example, people might want to write a NP node to publish a `/tf` topic if they don't already have one.

bunnyc1986, 2021-11-24:
thanks for posting the issue. I found the 3D panel automatically picks up the topic if its output type is `tf2_msgs__TFMessage`. So having the prefix doesn't affect transforms to be published from node playground.

amacneil, 2021-12-15:
@defunctzombie @jtbandes @jhurliman is there any reason we enforce this prefix? Why don't we let NP publish on any topic?

defunctzombie, 2021-12-15:
I don't think there's a technical requirement. For end users it might be nice to disambiguate between what studio generated locally and what came from their data source - but we don't have to impose that on folks.

amacneil, 2021-12-15:
Agree that it's a nice property, but let's leave that up to the node author. They can always add whatever prefix they want.

jtbandes, 2021-12-15:
No reason, other than sanity at Cruise when dealing with bags containing hundreds of topics.

defunctzombie, 2021-12-16:
@amacneil I think this is better for Jan 2022 roadmap given the other items and bug backlog we have.

jhurliman, 2022-01-07:
The main technical reason we do this now is to check for circular dependencies with node playground input/output topics. Any input topics that start with `/studio_node/` are considered invalid. Imposing this same restriction without the prefix will require a bit of code refactor.

defunctzombie, 2022-01-08:
@jhurliman is it correct that a node playground node cannot listen to the output of another node playground node? Or does that work today? So we'd only need to do a circular dependency check on the input/outputs of the currently active nodes.

jhurliman, 2022-01-08:
@defunctzombie I have no idea if this is just a circular dependency prevention or if piping node playground scripts together doesn't work at all.

PR #2990 (2022-03-08): Node Playground: remove "/studio_node" prefix requirement
**User-Facing Changes**
Removed the requirement for Node Playground node output topics to start with a "`/studio_node`" prefix.

**Description**
Resolves #2210
commit 7f939880 (2022-03-08): Node Playground: remove "/studio_node" prefix requirement (#2990)

**User-Facing Changes**
Removed the requirement for Node Playground node output topics to start with a "`/studio_node`" prefix.

**Description**
Resolves #2210
closed 2022-03-08 by jtbandes