foxglove/studio #2211: Hide "Add" button on "State Transitions" panel when pointer is not on the panel.

bunnyc1986, 2021-11-24:
Currently, the "add" button blocks part of the plot. Could you please make is hidden when mouse pointer is not on the panel like the common toolbar? 

[image]

jhurliman, 2022-01-07:
[image]

defunctzombie, 2022-01-09:
@2metres sending this bug your way. As a start we could match the plot panel behavior to hide the button unless hovered. I think in this case we might want to do better since when you hover this button covers important parts of the dataset.

Some ideas:

- Make state transition panel only support one dataset and the user opens multiple panels to show multiple datasets. Since we have timeseries pan/zoom sync this would still work.
- Add a button under the entire plot to add a series
- Whatever other magical idea you can think of for the add button

foxymiles, 2022-01-12:
[image]

How about making it an always visible button in the bottom right corner?

defunctzombie, 2022-01-12:
If you have a short panel then it will overlap your dataset.

foxymiles, 2022-01-12:
@defunctzombie Yeah I don't see anywhere we can put it where it will never overlap. Should I go ahead and hide it when not hovered? The "reset view" button has a similar issue.

PR #2674 (2022-01-14): Hide the state tranition panel add button when panel isn't hovered.
**User-Facing Changes**
This hides the add button on the state transition panel when the panel isn't hovered to prevent it from overlapping the chart.

**Description**
This just uses the `usePanelMousePresence` hook to hide & show the button.

Fixes #2211
closed 2022-01-14 by foxymiles