foxglove/studio #2212: Data transform ("node") extensions

jtbandes, 2021-11-24:
Some users have input data with schemas not directly compatible with Studio built-in panels. Node Playground can be used to transform the original data into Studio-compatible data types (such as TF and Markers) but the custom nodes are saved as part of the user's layout configuration.

It would be beneficial to decouple the authoring of these transformations from the layout, so the transforms can be reused across an organization and maintained by a team other than the end users of Studio. This could mean creating new extension APIs for providing "data transforms".

bunnyc1986, 2021-11-24:
thanks for posting this feature. I'd like to add another requirement: these transformer should only be executing and pulling upstream data when the output topics are consumed.

defunctzombie, 2021-11-29:
Some API ideas I played around with https://gist.github.com/defunctzombie/42a5478d7240044095192cee65d84cae regarding datatype transformations.

rgov, 2022-06-03:
I had a very similar request ("non-panel extensions"), and while I think the proposed API is elegant for many transformations, I'm thinking about how this might apply to a stateful video codec, and it might be too limited.

How about an extra flag to `PanelExtensionContext.publish()` that allows the message to be internal-only?

jameskuszmaul-brt, 2022-06-17:
Is the intention of this request to just be focused on writing extensions to do this, or has any thought been put towards something to enable more user-side transformations? I'm imagining a more advanced version of what is available in plots (https://foxglove.dev/docs/studio/panels/plot#math-modifiers) where conceptually studio would provide an interface where the user could write a javascript function that would take in an object for a given time-point and return what they want to be plotted/printed out/displayed/etc.? The idea being to easily enable people to, e.g., plot the difference between two fields, or to be able to turn the contents of a message into something more human-readable that could be displayed in the Raw Messages panel, without needing to deal with the overhead of writing your own extension (but also with more limitations).

rgov, 2022-06-17:
@jameskuszmaul-brt I think what you are describing is almost exactly what the [Node Playground](https://foxglove.dev/docs/studio/panels/node-playground) panel is.

jameskuszmaul-brt, 2022-06-17:
@rgov Ah thanks. I thought I'd seen something like that and then must've conflated that with extensions in my mind and assumed that only extensions were possible. Thanks!

jhurliman, 2022-09-22:
Related: https://github.com/foxglove/studio/pull/4456

jtbandes, 2022-12-13:
This was done in https://github.com/foxglove/studio/pull/4507. To learn more, see the announcement: https://foxglove.dev/blog/announcing-message-converter-extensions-in-foxglove-studio

closed 2022-12-13 by jtbandes