foxglove/studio #2214: Change datatypes API to include enum information

jtbandes, 2021-11-24:
Currently we use `RosDatatypes` throughout the app. This data structure can include enum constants, but each message field is not annotated with a type indicating which set of constants it comes from. The `__foxglove_enum` hack can be used to add annotations to ROS msg fields matching them with a separate message containing only constants.

Protobuf, however, has native support for enums, so data sources using protobuf schemas will have information about which fields are associated with which enums. The type of `datatypes` and the `enumValuesByDatatypeAndField` will need to change to take full advantage of protobuf enums.

jtbandes, 2022-04-01:
It turns out Protobuf may not need so much special handling (see https://github.com/foxglove/studio/pull/3130), because `enumValuesByDatatypeAndField` depends on the **order** of the fields and constants. As long as the constants are added to the list immediately before the field itself, the names are correctly picked up at least for this simple case.

It would still be good to address this directly in the schema so that we don't have to rely on this hacky ordering-based check.

defunctzombie, 2022-04-01:
Sounds like we should look at addressing this with this work:
https://github.com/foxglove/studio/pull/3091

Or at least writing up a high level issue that describes the path from `RosDatatypes` to a ros-agnostic schema that is capable of conveying this information.