foxglove/studio #2222: Strange error when attempting to install .deb for the wrong architecture

jhurliman, 2021-11-29:
**Description**

When attempting to install the amd64 deb on an arm64 architecture Ubuntu install, I get a strange error that doesn't hint at what the actual problem is:

```
$ sudo apt install ./foxglove-studio-0.22.1-linux-amd64.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'foxglove-studio:amd64' instead of './foxglove-studio-0.22.1-linux-amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 foxglove-studio:amd64 : Depends: libgtk-3-0:amd64 but it is not installable
                         Depends: libnotify4:amd64 but it is not installable
                         Depends: libnss3:amd64 but it is not installable
                         Depends: libxtst6:amd64 but it is not installable
                         Depends: libatspi2.0-0:amd64 but it is not installable
                         Depends: libdrm2:amd64 but it is not installable
                         Depends: libgbm1:amd64 but it is not installable
                         Depends: libxcb-dri3-0:amd64 but it is not installable
                         Recommends: libappindicator3-1:amd64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

I would expect to see this error instead: `(--install: package architecture (amd64) does not match system (arm64)`

defunctzombie, 2021-11-29:
This looks like an error from `apt` and not anything we can change. Have you seen any documentation or references indicating otherwise?

jhurliman, 2021-11-29:
[wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb.zip](https://github.com/foxglove/studio/files/7621066/wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb.zip)

Try unzipping and installing this .deb on an arm64 Ubuntu install using `sudo apt install ./wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb` and observe the following result:

```
$ sudo apt install ./wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb 
[sudo] password for jhurliman: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'wireless-bcm43142-dkms:amd64' instead of './wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb'
The following NEW packages will be installed:
  wireless-bcm43142-dkms:amd64
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/1,337 kB of archives.
After this operation, 4,262 kB of additional disk space will be used.
Get:1 /home/jhurliman/Downloads/wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb wireless-bcm43142-dkms amd64 6.20.55.19-1 [1,337 kB]
dpkg: error processing archive /home/jhurliman/Downloads/wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb (--
unpack):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 /home/jhurliman/Downloads/wireless-bcm43142-dkms_6.20.55.19-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

This packages correctly throws the `package architecture (amd64) does not match system (arm64)` whereas our .deb does not.

jhurliman, 2021-11-29:
This appears unfixable.

closed 2021-11-29 by jhurliman