foxglove/studio #2248: Toolbar disappears in fullscreen mode if the panel is in a tab

bunnyc1986, 2021-12-01:
**Description**
Toolbar disappears in fullscreen mode if the panel is in a tab

- Version: The latest version on demo website
- OS: Mac OS 11.6, and Chrome 96.0.4664.55

**Steps To Reproduce**

[image]

**Expected Behavior**
The toolbar should show up as if the panel is not in a tab.

**Actual Behavior**
Toolbar disappears in fullscreen mode if the panel is in a tab.

defunctzombie, 2021-12-03:
@bunnyc1986 could you elaborate on the steps to reproduce this?

I create a new tab panel and an image panel inside the tab panel. I can fullscreen both panels and continue to see the appropriate panel controls.

jtbandes, 2021-12-03:
Maybe this is a z-index issue? I see some problems when fullscreening a panel that's inside a tab if there are also other panels adjacent to the tab:

https://user-images.githubusercontent.com/14237/144651986-214ca4d3-b64a-41d2-a41d-fc38b3a0ce16.mov

bunnyc1986, 2021-12-03:
> @bunnyc1986 could you elaborate on the steps to reproduce this?
> 
> I create a new tab panel and an image panel inside the tab panel. I can fullscreen both panels and continue to see the appropriate panel controls.

@jtbandes 's screen recording has the exact reproduction steps. Thanks for posing it!

PR #2276 (2021-12-06): Toggle fullscreen state for parent panels
**User-Facing Changes**
Fullscreen toggle for panels works correctly for panels within tabs.

Removes the fullscreen option from the quick access overlay.

**Description**
When a panel within a tab enters fullscreen, it needs to tell the tab panel to enter fullscreen to properly appear on top of other panels and controls.
    
This change also removes the "fullscreen" feature from the quick access controls. There is already a fullscreen icon in the panel toolbar which is available without having to open any menus or special shortcut keys.
    
Fixes: # [... cut]
commit 06c0db96 (2021-12-06): Toggle fullscreen state for parent panels (#2276)

When a panel within a tab enters fullscreen, it needs to tell the tab panel to enter fullscreen to properly appear on top of other panels and controls.

This change also removes the "fullscreen" feature from the quick access controls. There is already a fullscreen icon in the panel toolbar which is available without having to open any menus or special shortcut keys.

Fixes: #2248
closed 2021-12-06 by defunctzombie