foxglove/studio #2250: Add a rolling average math modifier for plots

Ryan-Holben, 2021-12-01:
It would be very helpful to have a rolling average math modifier for plots.  For example, suppose I'm plotting runtime values published by a node.  That data can be very noisy, so overlaying this with a second plot of the rolling average can help with interpretability. 

I'm not sure what the best interface would be for specifying the averaging interval, but for example if we are doing time series data, it could look like:  `.@average{0.5}` to indicate a half second window.

jhurliman, 2021-12-01:
Would specifying a message count (`.average{5}` for a 10hz topic over a 0.5s window) work? This would be more generic than assuming a timestamp is available.

Ryan-Holben, 2021-12-01:
Yes, absolutely.

Ryan-Holben, 2021-12-01:
Though a downside is that assumes the topics come in at a regular cadence.  So there are tradeoffs.

closed 2022-07-01 by github-actions[bot]