foxglove/studio #2270: Stepping through a bag results in a different end state than playing a bag

jhurliman, 2021-12-03:
[markertest009.bag.zip](https://github.com/foxglove/studio/files/7652913/markertest009.bag.zip)

End state after playing back the bag to the end:
[image]

End state after pressing the right keyboard button until the end of the bag is reached:
[image]

jtbandes, 2021-12-03:
This is kind of by design. Stepping is really a seek, and seeking to the end can result in a different end state as well.

I think the ticket should specifically call out TFs or markers, because those are some of the only things the app displays in a truly stateful manner (where to display them correctly you technically need to process every preceding message in the entire bag)

amacneil, 2021-12-06:
From live discussion: Let's scope this to supporting "step forward in time" only (e.g. for seeks <100ms).

Seeking forward large distances, or stepping backwards can continue to treat this as a seek and throw away existing visualization state.

PR #2418 (2022-01-20): Forward seek plays all messages until the seek time is reached

**User-Facing Changes**
Seeking (via the keyboard shortcut or the playback control button) triggers playback to start and automatically stop when the datasource time passes the desired seek location. This allows a user to repeatedly forward seek and achieve the same result with visualization as playing the datasource. This functionality impacts panels which rely on message state (i.e. coordinate frames)  for proper visualization.

**Description**
Forward seek plays forward rather than jump seek
    
Rather than jumping forward in  [... cut]
commit da664961 (2022-01-20): Forward seek plays all messages until the seek time is reached (#2418)

Rather than jumping forward in the datasource a fixed amount, seeking forward plays the data until at least the additional amount of time has elapsed in the datasource.

This allows repeated forward seeking to achieve the same playback result as playing through the datasource start to finish.

Fixes: #2270
closed 2022-01-20 by defunctzombie