foxglove/studio #2278: Point cloud does not render from demo bag

defunctzombie, 2021-12-04:
**Description**
When I open the demo bag and select the `/velodyne_points` topic in a new 3d panel I expect to see the point cloud show up. However nothing shows up when I play the file.

Regression from: https://github.com/foxglove/studio/pull/2243

**Steps To Reproduce**

1. Open a new layout
2. Add a 3d panel
3. Open the demo bag
4. Select `/velodyne_points` in the 3d panel
5. Play the file

**Expected Behavior**
After the above steps I expect to see point clouds rendered in the 3d scene.

**Actual Behavior**
Nothing is rendered in the 3d scene.

jhurliman, 2021-12-05:
I cannot repro using either latest `main` or https://studio.foxglove.dev/ and `demo.bag`. Screenshots:
[image]
[image]

defunctzombie, 2021-12-06:
The issue is that when adding a new 3d panel, then selecting the `velodyne_points` topic - I should see the points show up. That does not happen. In the above screenshots you are looking at the welcome layout which is already configured. Try adding a blank 3d panel and then enabling the topic.

The behavior I am describing should show the selected topic without the user having to configure the frame.

defunctzombie, 2021-12-06:
Seems like if this is a regression it isn't recent. Closing out the bug and we can separately discuss any UX around making it more obvious where to select the frame.

closed 2021-12-06 by defunctzombie