foxglove/studio #2284: Map panel: add user-defined GeoJSON layer

jessicaaustin, 2021-12-06:
When viewing the Map Panel, it would be helpful to be to "annotate" the map, for example to mark a location of interest. 

One relatively quick way to do this would be to allow the user to specify a GeoJSON in the panel config. Then that could be loaded into the leaflet map. 

An example: 

Code: [add custom geojson layer ](https://github.com/jessicaaustin/studio/commit/a4349fda925d2d72e74b49856b9289466508ec7a)

In Studio:  [image]

jessicaaustin, 2021-12-06:
I attached a PR where I'm playing around with this idea.

If you guys are interested in this idea, and agree with this approach, I'd like to flesh this out and enable a user to specify the GeoJSON in panel settings. I'm not quite sure what the best way is to add panel settings for a panel that's using the PanelExtensionAdapter though -- could you point me in the right direction?

defunctzombie, 2021-12-07:
> I'm not quite sure what the best way is to add panel settings for a panel that's using the PanelExtensionAdapter though -- could you point me in the right direction?

I think we haven't exposed that to extensions - because we ourselves weren't too sold on how settings are setup today for non-extension panels.

Regarding displaying geoJSON data (or any overlays) is the use case to have a static file referenced or is the use case to have a topic that could produce geoJSON data and that data comes from the robot and could change over time?

jessicaaustin, 2021-12-07:
> Regarding displaying geoJSON data (or any overlays) is the use case to have a static file referenced or is the use case to have a topic that could produce geoJSON data and that data comes from the robot and could change over time?

The use case is static. Either a static file, or even have the user copy/paste in a block of geojson in the panel settings.

PR #3078 (2022-03-25): Support GeoJSON messages
**User-Facing Changes**
This adds support for a new `foxglove.GeoJSON` message type that adds GeoJSON markers to maps.

**Description**
This upgrades the map panel to subscribe to topics with the `foxglove.GeoJSON` datatype and to render them alongside the existing marker types.

[image]

Fixes #2284
closed 2022-03-25 by foxymiles