foxglove/studio #2299: load remote bag from url state fails

frapit, 2021-12-07:
**Description**
The recently introduced feature (https://github.com/foxglove/studio/pull/2171, thank you @foxymiles) to represent the current state (bag+layout+time) in the url misbehaves in my first tests. I always get redirected to the welcome scenario at 
`https://example.org/foxglove/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fstorage.googleapis.com%2Ffoxglove-public-assets%2Fdemo.bag&layoutId=bff4eafa-77d2-4c0f-87a1-a19731fa887b&time=2017-03-22T02%3A26%3A20.202843112Z` instead of the desired remote bag. 

I use the docker container for a self hosted version and the only adjustion I made is to host at a subpath `/foxglove`.

- Version: 0.23.0, self hosted
- OS: ubuntu20.04

**Steps To Reproduce**
In the start up view, load a new remote bag via the panel -> bag loads successfully -> url gets replaced.
Then copy that url and paste it into a new chrome session -> bag and view shows up for a second but then the welcome layout + bag loads and overwrite the state. 

Do I use it wrong or what could be the issue here?
Thanks

frapit, 2021-12-07:
Sure, I use the Dockerfile from the repo. I changed the `publicPath` property in `web/webpack.config.ts` to be `publicPath: "/foxglove/"` I build the container and launch using the following docker-compose.yml:

```
version: '3.6'
services:
    foxglove:
        image: foxglove:latest
        network_mode: host
        restart: unless-stopped
        volumes:
            - ./Caddyfile:/src/Caddyfile:ro
        command: ["caddy", "run"]
volumes:
  Caddyfile:
```

with the `Caddyfile` being:
```
:8080 {
	handle /foxglove* {
		root * ./
		try_files {path} /index.html
		file_server
	}
```
and some additional routing with traefik.

The same thing happens for local deployment and also the publicly hosted demo at [https://studio.foxglove.dev/](https://studio.foxglove.dev/) as well so I am not sure if it is related to the docker setup.

foxymiles, 2021-12-07:
Thanks for the extra details. Is it possible for you to share the exact URL you're seeing this happen with on studio.foxglove.dev?

frapit, 2021-12-08:
Maybe I was wrong and it is related to my setup. Not necessarily the foxglove but the bag hosting. Some more details:

My bags are hosted in a [bagdb](https://github.com/swri-robotics/bag-database) in a cooperate network, thats why I unfortunately can't share the link. 

On studio.foxglove.dev it does not load the bag, because of some cross-origin error when I go to the URL like the following `https://studio.foxglove.dev/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fexample.org%2Fbagdb%2Fbags%2Fdownload%3FbagId%3D4665&layoutId=3bc9ced1-efce-40d0-bf54-e3df85077d29`
Still, if I paste the link to a new session it gets replaced by the demo but I think that is understandable as the loading fails.

On my own hosted version it is a different story. Under "Connection" I choose "New remote bag (ROS 1)" then I paste the link of the bag: `https://example.org/bagdb/bags/download?bagId=3887` It is loaded successfully (I can stream images topics) and the url is being expanded to something like: `https://example.org/foxglove/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fexample.org%2Fbagdb%2Fbags%2Fdownload%3FbagId%3D3887&layoutId=f296b7cb-879b-447d-9030-390f2b9d1f08`

I now paste that link into a new chrome session and I am directed to the demo bag.
In the developer console I see
```
ExtensionLoaderProvider.tsx:16 [providers/ExtensionLoaderProvider.tsx] Fetching builtin extensions
useInitialDeepLinkState.ts:43 [../../packages/studio-base/src/hooks/useInitialDeepLinkState.ts] Error: Unknown path.
    at Lo (appURLState.ts:65)
    at useInitialDeepLinkState.ts:41
    at Object.useMemo (react-dom.production.min.js:172)
    at C.useMemo (react.production.min.js:23)
    at Zr (useInitialDeepLinkState.ts:37)
    at Qr (URLStateSyncAdapter.tsx:11)
    at Ga (react-dom.production.min.js:157)
    at Eu (react-dom.production.min.js:267)
    at hu (react-dom.production.min.js:250)
    at nc (react-dom.production.min.js:250)
(anonymous) @ useInitialDeepLinkState.ts:43
```
However, this does not happen If I use a different, publical available bag such as the webviz demo bag: `https://example.org/foxglove/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fopen-source-webviz-ui.s3.amazonaws.com%2Fdemo.bag&layoutId=74e58763-d268-4c93-9b98-963205eddbc3&time=2017-03-22T02%3A37%3A58.100878519Z` works fine in different sessions.

I am still wondering why the streaming from the bag database works initially but not in a new session.

Thanks you very much for your help!

foxymiles, 2021-12-08:
Thanks for your very detailed response. That helps a lot. I hadn't considered your use case of mounting the app under a subpath like `/foxglove` when I wrote the deep link handling code but that approach should work so I've addressed this in https://github.com/foxglove/studio/pull/2326.

frapit, 2021-12-09:
Thanks for the fix. I think the main issue is gone I don't get the error above anymore, and I can open bags via links from the bagdb. Yay! Still, when I loaded a bag and copy the URL into another chrome session, I get forwarded to the demo bag **at the first time**. If I enter the URL in this new session again, I will get the correct bag. In the network tab I get the following output:
[image]

Somehow both the bags are requested and the demo bag overrides the other.

foxymiles, 2021-12-09:
Thanks again for the extra details @frapit .

Can you try using this as a remote bag url and see if you get the same problem?

https://s3.amazonaws.com/www.plastaq.com/example.bag

frapit, 2021-12-09:
Yes, it does happen with your bagfile as well. No errors in the console, it"s loading the requested bag first and then the demo bag a second later and overwrites it. This happens only on a fresh chrome session (Inkognito Mode or with developer console and disabled cache.) and only on the first try. Subsequent tabs in the same session show the correct bag.

foxymiles, 2021-12-09:
Ah I think what's happening here is it's try to load the "welcome" session since it thinks it's a first time load.

frapit, 2021-12-10:
This matches my observation. If it is expected behavior, the issue can be closed I guess. Thanks for the fix!

amacneil, 2021-12-10:
I wouldn’t classify this as “expected behavior”, it’s just a bit of an edge case that we need to handle.

closed 2021-12-14 by foxymiles