foxglove/studio #2304: Decouple topics and datatypes

jtbandes, 2021-12-07:
The Player API currently requires players to produce:
- `topics: Array<{name: string, datatype: string}>`
- `datatypes: Map<string, RosMsgDefinition>`

This is built on the assumption that datatypes, and datatypes of nested complex fields, are uniquely identified by a string, across all topics. However, many container formats we support (including ROS 1 bags, mcap, and the new websocket protocol) provide fully self-contained schemas for each topic/channel, and it would make more sense for the data structures to reflect that. For example, we could change the Player API to be something more like `topics: Map<string, RosMsgDefinition>`.

defunctzombie, 2021-12-07:
Another thought would be `topics: Map<string, Schema>` where schema is a full AST of the message on the topic. Its like RosMsgDefinition but not ros specific.

Is there an underlying feature driving this rather than only tech debt? I'd like us to move away from tech debt only issues and tie them back to specific features that are blocked or not possible until resolved.

jtbandes, 2021-12-08:
The feature is to support data sources with schema name collisions or heterogeneous topics.

amacneil, 2022-12-16:
Still relevant?

jtbandes, 2022-12-17:
roman:
>  I'd like us to move away from tech debt only issues and tie them back to specific features that are blocked or not possible until resolved.

also roman:
> - https://github.com/foxglove/studio/issues/4895

jtbandes, 2022-12-17:
This weirdness in our code still exists. However we haven't gotten any feature requests for https://github.com/foxglove/studio/issues/2304#issuecomment-988398599 as far as I've heard.

closed 2022-12-17 by jtbandes